things4u/ESP-1ch-Gateway-v4.0 #1: how does CAD get enabled?

dlarue, 2017-02-06:
It is _CAD is defined as 1 in ESP-1ch-gway.h(#define _CAD 1 )  but looking at the console it states:

WiFi connected. local IP address: 192.168.1.208
Wlan Connected to MYSSID
Local UDP port 1700
Connection successful
MAC: 5c:cf:7f:c3:ae:a5:
Gateway ID: 5CCF7FFFFFC3AEA5, Listening at SF7 on 902.30 Mhz.
Time: Sunday 21:58:42
SSID=MYSSID
PASS=MYPASSWORD
FREQ=0
SF  =7
FCNT=0
DBUG=3
CAD =0
HOP =0
writeConfig:: done
Gateway configuration saved
Admin Server started on port 80
--------------------------------------
SX1276 detected, starting.

brainstorm, 2017-03-26:
@dlarue Have you tried enabling it through the "Admin Server web interface"? It works for me, yet it seems to be experimental since it crashes my gateway randomly when activated :/

dlarue, 2017-03-26:
I've not looked into it any further since.  There's a Lua version out there and it works well so that's what I've been running on the single channel WeMos gateway.

platenspeler, 2017-06-20:
So either use the admin webserver interface (it works MUCH better than Lua :-) ) or you use the ESP-sc-gway.h file where you can initialise all the settings you like

brainstorm, 2017-06-20:
I actually moved on to the Lua GW and it does work much better than this gateway code (much more stable, no reboots when CAD+HOP are enabled), @platenspeler :/

closed 2017-07-26 by platenspeler