things4u/ESP-1ch-Gateway-v4.0 #2: OTAA: downlink messages not showing up and OTAA not working

dlarue, 2017-02-06:
I have a node on a single channel(0) and SF10 and trying OTAA with this multi SF implementation and I'm not seeing downlink messages registering on the ESP's web page and after 1 hour no join happened.

FWIW, the Lua based CAD implementation takes less than 30 minutes to join and it does join/activate the node.

platenspeler, 2017-11-16:
Hi,

I\m working on a new release of the software for which the state machine is retritten. Downstream seems to work and OTA works (node I have at home). I'm doing some more tests and if successful I will publish the new code shortly.

platenspeler, 2017-11-18:
Please have a look at version 5 of the software and try again. I have a couple of 1-ch gateways in house and used them to both do OTA and downlink messages.
Mots of the times it works, but downlinks do not always work.

closed 2018-05-22 by platenspeler