things4u/ESP-1ch-Gateway-v4.0 #4: Response after PULL_RESP

mogyoros, 2017-03-08:
Dear Marteen!

I am using your code with an ESP12E and an SX1278 to build a gateway prototype. So far I am really happy with the operation and capabilities of the gateway.
For the backend, I am using the opensource lora server.
During the tests, when I was trying to send a **confirmed data** from the node (RN2483), the backend showed up an error:
```
"gateway: sending udp packet to gateway" addr=x.y.z.w:1700 protocol_version=1 type=PullResp`
"gateway: could not handle packet: gateway: unknown packet type: PullACK" addr=x.y.z.w:1700 data_base64="AQAABA=="
```
Digging into the protocol and your coode, I have found that you are replying with PULL_ACK message for a downstream PULL_RESP, which is leading to this issue. Per protocol you should reply with an TX_ACK message.

Therefor I have modified the **loraModem.h** and added PKT_TX_ACK definition:
`#define PKT_TX_ACK          0x05`

Plus had to change the code in ESP-sc-gway.ino in the readUdp function to adapt it to the TX_ACK message to be sent to the server:
```
case PKT_PULL_RESP:	// 0x03 DOWN

		lastTmst = micros();					// Store the tmst this package was received
		// Send to the LoRa Node first (timing) and then do messaging
		if (sendPacket(data, packetSize-4) < 0) {
			return(-1);
		}
		// Now respond with an PKT_PULL_ACK; 0x04 UP
// Mogyi: Changed from PULL_ACK to TX_ACK 0x05 UP, to be compliant with the protocol
		buff_up[0]=buff_down[0];
		buff_up[1]=buff_down[1];
		buff_up[2]=buff_down[2];
//              buff_up[3]=PKT_PULL_ACK;
                buff_up[3]=PKT_TX_ACK;
// Mogyi: And added the gateway MAC to the message to be sent
                buff_up[4]  = MAC_array[0];
                buff_up[5]  = MAC_array[1];
                buff_up[6]  = MAC_array[2];
                buff_up[7]  = 0xFF;
                buff_up[8]  = 0xFF;
                buff_up[9]  = MAC_array[3];
                buff_up[10] = MAC_array[4];
                buff_up[11] = MAC_array[5];
		buff_up[12]=0;
		
		// Only send the PKT_PULL_ACK to the UDP socket that just sent the data!!!
		Udp.beginPacket(remoteIpNo, remotePortNo);
		if (Udp.write((char *)buff_up, 12) != 12) {
//			Serial.println("PKT_PULL_ACK:: Error writing Ack");
                        Serial.println("PKT_TX_ACK:: Error writing TX ACK");
		}
		else {
			if (debug>=1) {
//				Serial.print(F("PKT_PULL_ACK:: tmst="));
                                Serial.print(F("PKT_TX_ACK:: tmst="));
				Serial.println(micros());
			}
		}
		//yield();
		Udp.endPacket();
		
		if (debug >=1) {
			Serial.print(F("PKT_PULL_RESP:: size ")); Serial.print(packetSize);
			Serial.print(F(" From ")); Serial.print(remoteIpNo);
			Serial.print(F(", port ")); Serial.print(remotePortNo);	
			Serial.print(F(", data: "));
			data = buff_down + 4;
			data[packetSize] = 0;
			Serial.print((char *)data);
			Serial.println(F("..."));
		}
```

With this change I could remedy the issue, and now the error message is gone, instead I see the receipt of the TX ACK message confirmation in the server log:
```
"gateway: sending udp packet to gateway" addr:x.y.z.w:1700 protocol_version=1 type=PullResp
"gateway: received udp packet from gateway" addr=x.y.z.w:1700 protocol_version=1 type=TXACK
"gateway: tx ack received" mac=5ccf7fffff1e22fe random_token=0

```

If you find it useful, you can interpret into your code aswell. And hope helps the others aswell.
Regards,
Tamas

platenspeler, 2017-03-21:
Thanks!

I'm working at this project in spare and vacation time, so I will review and include the code shortly when I have time to work on the single channel again.

Maarten

mogyoros, 2017-03-28:
Dear Maarten!
To support you I have created a pull request to adopt the changes, together with an another small modification.
Cheers,
Tamas

platenspeler, 2017-06-22:
Thanks!!

I worked today on a new version 4.0.3 in which I solved a lot of smaller issues, and I have incorporated the changes you suggested. Thanks again! Hope you see all your proposed changes implemented.

Changes include new webserver build which output the page in chucks and thus saves a lot of memory. The result is tha the system works much more reliably. And I made the page refresh every 60 seconds so we keep updated on packages coming in. 
Also I added cmac functionality so that the sensor emulation works as expected.

I will focus on HOP functions next few days.

Cheers,

Maarten

mogyoros, 2017-06-27:
Dear Maarten!

Checked the code, and yes, the changes were incorporated into the new code.

During the last 3 months, I was heavily on to test and develop your code, so I made several changes affecting for example the date&time printing, the gateway ID handling, OTA upgrade, gw statistics handling and stats update, gateway UDP management, removing some local variables and using direct function calls to save memory and an Uptime print function to the webpage (because I am too lazy to do math myself Current Time - Alive since...).

All these changes resulted in the gateway operation to be more stable, (no more ESP exceptions seen!), and an uptime of max 2 weeks.

If you are interested on the above, I can share the details.

However what I could not find out, is that sometimes the radio module gets stuck in a state, where no interrupts are generated, so packet forwarding stops. Resetting the module (via gateway UDP management), or even just changing the frequency forth and back (for example via the webserver) also triggers the radio module to escape from this state. I was heavily trying to analyze the state machine, but the devil is still hiding there...

Do you see such behaviour aswell?
Cheers,
Tamas

platenspeler, 2017-07-23:
Tamas,

Thanks you for your comments. Indeed the radio sometimes shows strange behavior and using CAD on/off seems to help, I have put a new version of the code on the webpage and this does not seem to happen anymore.

This version also has OTA and shows more information on the webpage. It also renews the page if you want (works on most browsers) and memory has increased a lot since I reworked the wwwServer.ino file. 

I am interested in your code, especially to see if you found any other stuff that needs enhancing.

mogyoros, 2017-09-23:
Hi Marteen!
Sorry not coming back earlier, I got a new job, so did not had time to focus on the project as earlier.
Thanks for the changes, specially the web server enhancement part. I will look at it in the coming weeks, and let you know my thoughts/findings.

Since I've made a lots of changes in the code, and did not sync it back to my fork (I was not really up to date, how to use github at that stage) it would be a real mess now to sync it. 

Hereby I share my "enhancements":

**GWID**
In every upstream packet you construct the UDP packet containing the GWID with this method, for example in pullData()
```
    pullDataReq[4]  = MAC_array[0];
    pullDataReq[5]  = MAC_array[1];
    pullDataReq[6]  = MAC_array[2];
    pullDataReq[7]  = 0xFF;
    pullDataReq[8]  = 0xFF;
    pullDataReq[9]  = MAC_array[3];
    pullDataReq[10] = MAC_array[4];
    pullDataReq[11] = MAC_array[5];
```
  Instead of this method I created a new variable containing the GWID with FFFF in the middle, and copying this variable to the packet:
```
// Global variable
uint8_t GW_ID_array[8];                 // Mogyi: Store gateway ID, MAC address + 2xFF in the middle

// For example in pullData()

// Mogyi: Removed local variable declarations
//	  uint8_t token_h = (uint8_t)rand(); 						// random token
//    uint8_t token_l = (uint8_t)rand();						// random token
    pullDataReq[1]  = (uint8_t)rand();            // random token
    pullDataReq[2]  = (uint8_t)rand();            // random token
    pullDataReq[3]  = PKT_PULL_DATA;			  			// 0x02
    memcpy(&pullDataReq[4],GW_ID_array,sizeof(GW_ID_array)); // Mogyi: modified to include 8 byte GW ID in the packet

// setup() part
    int k=0; 
    for (int i=0; i < sizeof(GW_ID_array); ++i){
        if ( i == 3 || i == 4){
            GW_ID_array[i]=0xFF;                              // Extend WIFI MAC with 2xFF
        } 
        else {
            GW_ID_array[i]=MAC_array[k];
            k++;
        }
    }

```

**GW STATS**

To store gw statistics packets I created a new struct:

```
// loramodem.h
struct gwstat_t {                       // Mogyi: gateway packet statistics 
    unsigned long tmst;
    uint8_t pkt_rxnb;                   // Mogyi: max 255 packets/period. Period is 1 min
    uint8_t pkt_rxok;
    uint8_t pkt_dwnb;
    uint8_t pkt_txnb;
} gwstat_t;
struct gwstat_t gwstats[MAX_STAT];      // Mogyi: to hold packet statistics records

```
Then I update this array's record 0 with the actual values in _loraModem.ino. For example:
```
gwstats[0].pkt_rxnb++;						              	// Receive statistics counter
```

When statistics interval is met in loop(), send the stats to network server, then shift the array, to hold the historical data until MAX_STATS

```
// ESP-sc-gway.ino
  	gwstats[0].tmst = millis();
    if (gwstats[0].tmst - stattime >= _STAT_INTERVAL) {		// Wake up every xx seconds
        sendstat();										// Show the status message and send to server
        for (int i=MAX_STAT-1; i>0; i--) gwstats[i]=gwstats[i-1];
        gwstats[0].pkt_rxnb = 0;
        gwstats[0].pkt_rxok = 0;
        gwstats[0].pkt_dwnb = 0;
        gwstats[0].pkt_txnb = 0;
     }
```
Of course in sendstat() need to accomodate the string assembly function:
```
    int j = snprintf((char *)(status_report + stat_index), STATUS_SIZE-stat_index, 
    "{\"stat\":{\"time\":\"%s\",\"lati\":%s,\"long\":%s,\"alti\":%i,\"rxnb\":%u,\"rxok\":%u,\"dwnb\":%u,\"txnb\":%u}}", 
    stat_timestamp, clat, clon, int(gwayConfig.alt), gwstats[0].pkt_rxnb, gwstats[0].pkt_rxok, gwstats[0].pkt_dwnb, gwstats[0].pkt_txnb);
```
And if you want you can display historical packet stats on the webpage
```
//_wwwServer.ino
if (debug >= 1) {
  response +="<h2>Packet statistics</h2>\r\n";
  delay(1);
  response +="<table class=\"config_table\">\r\n";
  response +="<tr>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">Time</th>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">LoRa packets received</th>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">LoRa packets OK</th>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">LoRa packets CRC error</th>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">Downlink packets received</th>\r\n";
  response +="<th class=\"thead\" style=\"width: 80px;\">Downlink packets emitted</th>\r\n";
  response +="</tr>\r\n";
  for (int i=0; i<MAX_STAT; i++) {
      if (gwstats[i].tmst!=0) {
          response +="<tr>\r\n";
          response +="<td class=\"cell\">"; response += printDateTime(gwstats[i].tmst,0)+" UTC"; response+="</td>\r\n";
          response +="<td class=\"cell\">"; response += gwstats[i].pkt_rxnb; response +="</td>\r\n";
          response +="<td class=\"cell\">"; response += gwstats[i].pkt_rxok; response +="</td>\r\n";
          response +="<td class=\"cell\">"; response += gwstats[i].pkt_rxnb-gwstats[i].pkt_rxok; response +="</td>\r\n";
          response +="<td class=\"cell\">"; response += gwstats[i].pkt_dwnb; response +="</td>\r\n";
          response +="<td class=\"cell\">"; response += gwstats[i].pkt_txnb; response +="</td>\r\n";
          response +="</tr>\r\n";
      }
  }
  response +="</table>\r\n";
```

Hope you find these useful for yourself too... If something is not clear please let me know, I will try to explain it more deeply.
Cheers & have a nice weekend!
Tamas