things4u/ESP-1ch-Gateway-v4.0 #10: connection issues without wifiManager

marcobrianza, 2017-06-29:
Hi, first of all thanks for this great project.
I have an issue connecting to wifi when the wifimanager is disabled.
Network credentials are populated correctly and used to work in version 2.0 of the gateway

Cinezaster, 2017-07-12:
Same here. it does work when I have the wifimanager enabled and I manually set the SSID and pass.

platenspeler, 2017-07-23:
Can you please send me the content of the wpas structure?

Cinezaster, 2017-07-24:
```
struct wpas {
	char login[32];
	char passw[64];
};

wpas wpa[] = {
	{ "xxxx" , "xxxxx" },
	{ "", ""},
	{ "", ""},
	{ "", ""}
};
```

platenspeler, 2017-07-24:
Ah, You should leave the first line empty like in the example. Start with your SSID/Password combinations as the second and more lines! I'll update the documentation accordingly. The first line is forWifiManager. If you do not use WifiManager it will start with te second line.

Cinezaster, 2017-07-24:
@platenspeler I'll try it tomorrow.

platenspeler, 2017-07-24:
regarding the documentation: I see that I included the warning already in the ESP-sc-gway.h file.

Cinezaster, 2017-07-25:
Ok thx, it does work. I used an older version.
I thinks this can be closed.

closed 2017-07-26 by platenspeler