things4u/ESP-1ch-Gateway-v4.0 #11: wemos reset itselt

jfmateos, 2017-07-14:
Dear Maarteen,
Thank you very much for sharing this project.
I am trying to configure a single channel gateway using your version 4 code, but something is not working properly.

First I tried with WifiManager enabled without success.
Now I have disabled WifiManager and hardcoded the ssid and password of the wifi AP, but the wemos resets itself (see image).
https://s02.justpaste.it/files/justpaste/d475/a17167248/file1.png
Please, could you help me to solve this problem?

Kind regards from Madrid

jfmateos, 2017-07-14:
Ok, I have managed to solve this.
The problem is that the pins definition inloraModem.h is for Hallard's board, not for the instruction I found at http://things4u.github.io/HardwareGuide/Single-Channel-Gway/ESP-gateway-v4.html

// Definition of the GPIO pins used by the Gateway
//struct pins {
	uint8_t dio0=15;	// GPIO15 / D8. For the Hallard board shared between DIO0/DIO1/DIO2
	uint8_t dio1=15;	// GPIO15 / D8. Used for CAD, may or not be shared with DIO0
	uint8_t dio2=15;	// GPIO15 / D8. Used for frequency hopping, don't care
	uint8_t ss=16;		// GPIO16 / D0. Select pin connected to SPIO16 / D0
	uint8_t rst=0;		// GPIO0 / D3. Reset pin not used	
// MISO 12 / D6
// MOSI 13 / D7
// CLK 14 / D5
// SS 16 / D0
//} pins;

// For ComResult gateway use the following settings
struct pins {
	uint8_t dio0=5;		// GPIO5 / D1. For the Hallard board shared betweenDio0/DIO1/DIO2
	uint8_t dio1=4;		// GPIO4 / D2. Used for CAD, may or not be shared with DIO0
	uint8_t dio2=0;		// GPIO0 / D3. Used for frequency hopping, don't care
	uint8_t ss=15;		// GPIO15 / D8. Select pin connected to SPIO16
	uint8_t rst=0;		// Reset pin not used	
} pins;

platenspeler, 2017-07-23:
Sorry for not responding sooner. Ny now you will have discovered that the pin definitions are in loraModem.h file, You should enable the standard pin setting setting by commenting the Hallard section and uncommenting the Comresult setting.
I'm thinking how to make the gateway autodiscover but I do not think this is possible.
I have updated the Read.me file to document how to get started with teh single channel gateway, which should in any case mean that you have to set the pins in loraModem.h, set the SSID and password for  your network in ESP-sc-gway.h and from that moment on cofnigure the gateway further.

closed 2017-08-06 by platenspeler