things4u/ESP-1ch-Gateway-v4.0 #12: Using ESP32

aizukanne, 2017-07-27:
Hello, Can this be implemented using ESP32 without modifications to the code?
Also can this work on 433MHz if the relevant changes are made to set frequency in the 433MHz range?

aizukanne, 2017-08-27:
I tried this on ESP32 without success. Lots of modifications were needed. Different libraries from ESP8266. Hope someone can MOD this for ESP32 soon.

platenspeler, 2017-09-12:
I bought one, and will try to build one myself.

aizukanne, 2017-09-21:
I have been able to implement a single channel gateway on the ESP32 and with the RFM95W LoRa Module. However i modified Pycom's code for their Lopy, inserting libraries for the SX1276 and modifying the code to achieve this. If interested, the source code is available from my github here https://github.com/aizukanne/micropython-lora.

There are a few issues though. I am not sure tx works. I probably commented out some lines i shouldn't have.

vshymanskyy, 2017-12-05:
I found a mod for ESP32 here: https://github.com/kersing/ESP-1ch-Gateway-v5.0

kb9nate, 2017-12-27:
I tried the link vshymanskyy posted but can not get it to compile, has anyone had luck with it and if so what did you have to do to get it to work?
I

platenspeler, 2018-06-02:
Version 5.2.0 supports most of the functions of the Single Channel Gateway. Please swith to version 5.

closed 2018-06-02 by platenspeler