things4u/ESP-1ch-Gateway-v4.0 #16: US frequency (915MHz) configuration

cesschneider, 2017-08-03:
Hello everybody, 

First of all, congratulations for this great work!

I'm running some tests with RFM95W (SX1276) chips and 915MHz antenas in Brazil, which is the adopted frequency here, and need some help to change some values at file "loraModem.h" in order to test on that frequency.

// what value should use here?
#define FRF_MSB						0xD9		// 868.1 Mhz

// are these freqs right?
int freqs [] = {
  915000000,
  902300000,
  903080000,
  905240000,
  907400000,
  909560000,
  911720000,
  913880000,
  916040000,
  918200000,
  920360000,
  922520000,
  924680000,
  926840000
}

uint32_t  freq = freqs[0];
uint8_t	 ifreq = 0;								// Channel Index

Appreciate any thoughts and help!
Best regards,

Cesar.

marciolm, 2017-08-15:
Hello Cesar
For while, this is a single frequency gateway, only the first one in the list is used. I use with TTN 903900000 or 916800000, not 915000000 as in your code.
For the 915 MHz band, you should  also change the FSR registers in loramodem.h file, like this example.

// FRF
//#define FRF_MSB						0xD9		// 868.1 Mhz
//#define FRF_MID						0x06
//#define FRF_LSB						0x66

#define FRF_MSB           0xE1    // 916.8 Mhz /61.035 = 15007782 = 0xE50026 or
#define FRF_MID           0xF9     // 903.9 MHz /61.035 = 14809535 = 0xE1F9BF
#define FRF_LSB           0xBF

Change the SF to SF7 too.

It's working for me, I receive uplink messages in TTN, but downlink messages aren't delivered to node, even with the strict mode set to 1.
Márcio

lagunacomputer, 2018-02-18:
hi were these the only code changes you had to make to get 915mhz working?

lagunacomputer, 2018-02-19:
ok i got it working, i had to delete all of my TNN gateways and apps, and recreate new ones before it worked

platenspeler, 2019-11-12:
I hope that with new version 6 support for all other territories does work.

closed 2019-11-12 by platenspeler