things4u/ESP-1ch-Gateway-v4.0 #17: Can't compile for WeMos D1 mini

JenertsA, 2017-08-06:
compiling returns **.text will not fit in region iram1_0_seg** error 
googoling all day but can't find anything that would solve problem.

`Archiving built core (caching) in: C:\Users\jener\AppData\Local\Temp\arduino_cache_648627\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_0261c50b50c09a0d4455295da7443622.a
c:/users/jener/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\jener\AppData\Local\Temp\arduino_build_379453/ESP-sc-gway.ino.elf section .text will not fit in region iram1_0_seg
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board WeMos D1 R2 & mini.`

JenertsA, 2017-08-06:
todays 4.0.8. version solves problem.

closed 2017-08-06 by JenertsA

platenspeler, 2017-08-06:
Thanks for elling me!