things4u/ESP-1ch-Gateway-v4.0 #20: OTAA not working after upgrading from 4.0.4 to 4.0.8

Cinezaster, 2017-08-25:
I was previously using version 4.0.4
I upgraded to version 4.0.8

I have one node in ABP mode (Nodemcu + RFM95W combination) working perfect
I have one node in OTAA mode (LoPy) I only see the  join messages in The TTN console but it never joins

All settings are the same. as far as I can tell.

platenspeler, 2017-08-26:
Hi,

I wil have a look one of these days. I did not (re) test the functions in release 4.0.8
I do have about 8 single channels working and two full blown gateways, so proper testing of the OTAA functions is a little more difficult since the full gateways will always respond ☺

Regards,

Maarten

Sukuri, 2017-12-05:
Hi,

Can I know how to make the gateway ABP mode?
just switch '#define A_OTA 1' to '#define A_OTA 0' ?

closed 2018-05-22 by platenspeler