things4u/ESP-1ch-Gateway-v4.0 #29: internal sensors

sprokkie, 2018-12-05:
Hi, got the gateway running with a lorago.
now i want to read the internal sensors.

i added the device the appskey and nwkey the device is recognized.
i cannot see the sensor data, do i need to configure more ?

how do i convert the payload to something i can read ?
now i got in the gatwway this payload : 400F16012600CF0001BFA6575E92AEB7
in the device data: 86803F

please help or push me in the right direction

platenspeler, 2019-11-12:
Please use version 5 or better version 6 of the code. Version 4 is obsolete and will be removed.
I hope that with later version you will be able to read the sensors.

closed 2019-11-12 by platenspeler

sprokkie, 2019-11-12:
Hi Maarten,

 

Update naar versie 6  moet ik alles dan opnieuw invullen of kan ik bestaande file’s houden ?

 

Groeten,

Ferry

 

Closed #29 <https://github.com/things4u/ESP-1ch-Gateway-v4.0/issues/29> .

platenspeler, 2019-11-12:
Je moet, omdat zo goed als ALLE files zijn geraakt alles opnieuw installeren. De parameter files zijn ook geraakt: ESP-1ch-gway.h is bijv. renamed als configGway.h maar er is ook een configNode.h file.
Ik denk overigens dat het aanpassen van de configuratie in versie 6 niet veel tijd kost omdat de velden zo goed als geliijk zijn.

Maarten

sprokkie, 2019-11-12:
Hoi Maarten,

 

Bedankt voor je antwoord.

Ik ben bang dat ik veel meer moet aanpassen.

Ik heb de lorago dock waar ik een door Sadnbox de producent aangepaste versie van jouw code gebruik

https://sandboxelectronics.com/?product=lorago-dock-single-channel-lorawan-gateway

https://github.com/SandboxElectronics/LoRaGoDOCK-Gateway

 

“LoRaGo DOCK - Single Channel LoRaWAN Gateway based on ESP8266 and SX1276

 

Version : 0.8.2

Date : 2018-11-26 Software: https://github.com/SandboxElectronics/LoRaGoDOCK-Gateway

Hardware: LoRaGo DOCK – http://sandboxelectronics.com/?product=lorago-dock-single-channel-lorawan-gateway

 

Originally designed by Maarten Westenberg (mw12554@hotmail.com)

Adopted by Sandbox Electronics (http://sandboxelectronics.com)

 

This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.”

 

Ik moet nu dus gaan uitzoeken wat ze precies hebben aangepast.

OTA werkt niet dus ik zal ook het dak op moeten de gateway naar binnen halen om te verbinden met USB :(

 

Met vriendelijke groet,

Ferry Meijlink

platenspeler, 2019-11-13:
Hoi Ferry,

De meeste files zijn voornamelijk van naam veranderd en zijn van een ander serienummer voorzien.
Helaas is de pin-out wel veranderd en is de #define is uit config.h file gehaald.
Maar in de nieuwe versie kunt je eenvoudig een nieuwe pin-out toevoegen, en de rest lijkt hetzelfde.

Mijn vermoeden is dat al we de pin-out aanpassen (nr. 6 wordt het dan) in de configGway.h file dat het dan moet werken. Er zijn meen ik geen andere zaken aangepast maar wel geconfigureerd.

Ook moet je de OTA aanzetten in de configuratie files, dan hoef je het dak niet meer op. Maar dat staat in principe al goed.

Maarten

Ps. Wel jammer dat sandbox mij niet een versie heeft gestuurd :-)
Ps2. Je kunt e.e.a al voorbereiden en compileren zonder het dak op te gaan.

sprokkie, 2020-01-06:
Hoi Maarten,

 

Ondertussen de gateway geüpgraded naar V6 en ik heb er nog een die draait ook op V6

Heb de interne sensoren nog niet aan de gang gekregen, zet ze wel aan maar de waarde is 0 wat er wordt overgestuurd.

 

Een andere vraag,

Waarom is een 8 channel gateway zo lastig te maken ?

Heb je daar andere hardware voor nodig ?

 

Kan je niet 8 singlechannels lora printjes aansluiten op een esp en dan iedere printje en ander kanaal toewijzen ?

Weer alleen even niet hoe je dat allemaal op 1 antenne moet aansluiten en anders 8 antennes die een stukje uit elkaar staan :)

 

Groeten,

Ferry